stm32duino/BoardManagerFiles #1: '-' in .build.board cause compiler warning

RickKimball, 2016-09-15:
```
/home/kimballr/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -Os -Wall -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD "-I/home/kimballr/Arduino/hardware/stm32/STM32F1/system/libstm32f1" "-I/home/kimballr/Arduino/hardware/stm32/STM32F1/system/libstm32f1/include" "-I/home/kimballr/Arduino/hardware/stm32/STM32F1/system/Drivers/STM32F1xx_HAL_Driver/Inc/" "-I/home/kimballr/Arduino/hardware/stm32/STM32F1/system/Drivers/CMSIS/Include/" "-I/home/kimballr/Arduino/hardware/stm32/STM32F1/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include" -mcpu=cortex-m3 -DARDUINO=10612 -DARDUINO_NUCLEO-F103RB -DARDUINO_ARCH_STM32F1  -mthumb -DSTM32F103xB "-I/home/kimballr/Arduino/hardware/stm32/STM32F1/cores/arduino" "-I/home/kimballr/Arduino/hardware/stm32/STM32F1/variants/STM32F103RB-Nucleo" "/home/kimballr/Arduino/hardware/stm32/STM32F1/cores/arduino/new.cpp" -o "/tmp/arduino_build_402627/core/new.cpp.o"
**<command-line>:0:15: warning: ISO C99 requires whitespace after the macro name [enabled by default]**
```
You might want to change in boards.txt:
NUCLEO-F103RB.build.board=NUCLEO-F103RB
to
NUCLEO-F103RB.build.board=NUCLEO_F103RB

rogerclarkmelbourne, 2016-09-25:
This is now fixed in the Arduino_Core_STM32F1 repo (using Rick's change) , but will not get propagated into the Boards manager package until the next stable release.

fpistm, 2017-01-19:
Issue take in account in the further package and core repo.

closed 2017-01-19 by fpistm