stm32duino/BoardManagerFiles #2: nucleoFlasher path and target (windows)

madias123, 2016-09-15:
As written here:
http://stm32duino.com/viewtopic.php?f=16&t=1398&p=17953#p17953
Under Windows I had to change  2 things:
platforms.txt
# Uploader tool
# -------------------
tools.nucleoFlasher.path={runtime.hardware.path}/tools/nucleoFlasher

and (as workaround)
 line 8 of nucleoFlasher.bat
VOL %%I: 2>NUL | FIND "NUCLEO" >NUL && SET DEST=%%I:

I installed the latest ST-Link 2.1 driver, updated the firmware

rogerclarkmelbourne, 2016-09-15:
I think there may be better ways to find the Nucleo device, rather than iterating through all available drives

Windows has a built in command called WMIC which has a pseudo SQL language that allows searching for drives of a specific name

`WMIC LOGICALDISK where "volumename like 'Nucleo%'" get deviceid,Volumename`

In a batch file this has to be changed to

`WMIC LOGICALDISK where "volumename like 'Nucleo%%'" get deviceid,Volumename`

However, I have not managed to pipe the output of this command to a variable in a batch file, because the command outputs 3 lines of text, and the second line contains the drive letter, (the first line is a header and the last line is empty)

I think its technically possible to extract the second line, - there are examples of how to do this on StackOverflow, but I have not managed to get any of them to work.

rogerclarkmelbourne, 2016-09-16:
Thanks

I don't have a Nucleo either, but I do have a different board that has the same sort of USB Mass storage upload

So I can try it on that

rogerclarkmelbourne, 2016-09-16:
Thanks

It works perfectly

(I am sure some other projects which use mass storage would also be interested in how to do this)

BTW.  I have now uploaded the sources to separate folders , so I should probably replicate at least part of this issue in the tools repo
https://github.com/stm32duino/Arduino_Tools

fpistm, 2017-01-19:
I think this issue could be close?

rogerclarkmelbourne, 2017-01-20:
Yes. I think I fixed this ages ago.

closed 2017-01-20 by rogerclarkmelbourne