stm32duino/BoardManagerFiles #4: Suggested change to boards section heading in platform.txt

rogerclarkmelbourne, 2016-09-15:
In platform.txt (for F103) the name for the boards supported by this file (really by boards.txt), is defined as

`name=STM32F1xx Boards`

If you look at the Arduino.cc  boards, they use

`name=Arduino ARM (32-bits) Boards`

I think this should probably be changed to

`name=STMicroelectronics STM32F1xx Boards`

(Same applies to the L4 etc)

I checked and this width of text is fine as the Arduino SAMD boards have a much larger name

rogerclarkmelbourne, 2016-09-25:
As we will support non STM branded boards, perhaps I will leave this as it is.

fpistm, 2017-05-18:
I think It could be closed as now it mentions core (MCU) and not boards:
"name": "STM32 Cores",

closed 2017-06-14 by fpistm