stm32duino/BoardManagerFiles #7: File not found

scargill, 2017-08-03:
VERY frustrating..  Ok, I started from SCRATCH - Arduino 1.8.3 - grabbed the URL, installed STM32F1xx Cores by ST MicroElectronics version 2017.1.20  - so everything up to date on Windows 10 64 bits.

I loaded a simple program - lighting the green onboard LED on the BLUEPILL F103C8 board.... alll well  - program compiled - via serial it loaded up on the board.  The light came on.

I made a minor change and..

arm-none-eabi-gcc: error: C:\Users\Peter\AppData\Local\Temp\arduino_build_514602/core/syscalls_stm32.c.o: No such file or directory

exit status 1
Error compiling for board BluePill F103C8.

I moved the Arduino directory from Program File x86 to simply "\Arduino" - as some old post suggested an issue with spaces  (note that some slashes are forward, some are back). Voila - compilation worked.

Made a change back... tried to compile again - back to square 1 !!!!

HELP.

fpistm, 2017-08-04:
Hi @scargill,
Merge of the [Arduino_Core_STM32F1](https://github.com/stm32duino/Arduino_Core_STM32F1) is on going into the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32) (generic core for all stm32 based board). See [Merge Arduino_Core_STM32F1 #20](https://github.com/stm32duino/Arduino_Core_STM32/issues/20)
I think, your issue comes from a new Arduino IDE option, open the _Preferences_ dialog box and in the _Setting_ tabs uncheck the _Aggressively cache compiled core_.
This issue is fixed in the new core. When merge of the  [Arduino_Core_STM32F1](https://github.com/stm32duino/Arduino_Core_STM32F1) will be done this core will be deprecated.

P.S.: please for any issue around cores submit issue on the related github. For this one: [Arduino_Core_STM32F1/issues](https://github.com/stm32duino/Arduino_Core_STM32F1/issues)
This one is mainly for installation issue thanks the BoardsManager.

scargill, 2017-08-04:
Much appreciated. Yes realise I should have used the other repository - and have done for an important i2c question. I'll try your mod, thank you.

fpistm, 2017-08-04:
I've reproduced the issue then disable the cache core and it's ok.

closed 2017-08-04 by fpistm