stm32duino/BoardManagerFiles #9: Urgently need Blue Pill support

dackley1935, 2017-09-03:
I'm presenting "How Program STM32F103C8/Blue Pill Using Arduino IDE" next Saturday (Sept 16)  to the Dallas Personal Robotics Group (dprg.org) where I had planned to demonstrate installing the STM32F1 cores package that supports the Blue Pill.

My problem is that the STM32F1 cores package is no longer available and the STM32 cores package doesn't yet support the Blue Pill.

fpistm, 2017-09-03:
Hi, @dackley1935,
You can use the previous commit of the json file  which includes the old STM32F1 core 

> https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/d1c46284dec37305d563a87eecfb545e3c66a166/STM32/package_stm_index.json

Or download it on your local machine then in  _Preference_ set:
`file:\<path on yourlocal machine\>\package_stm_index.json`

dackley1935, 2017-09-03:
Thanks you very much for the help. I really appreciate the quick response.

fpistm, 2017-09-03:
You're welcome ;)

closed 2017-09-03 by fpistm