stm32duino/BoardManagerFiles #10: Arduino 1.8.4 not showing added boards

Airheadbit, 2017-09-07:
I 'upgraded' to 1.8.4 and then added https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json 
to the preferences.txt file with the IDE, inside of board manager there is no STM libraries to load, nothing shows up.

fpistm, 2017-09-07:
Hi @Airheadbit,
It should be available in the 'contributed' section.
Which OS you use?
In the Arduino console have you got some error message? 

I'm running also with 1.8.4 on Windows and Linux and I do not have any issue.

fpistm, 2017-09-14:
Any news @Airheadbit?

Testato, 2017-09-14:
I think he is confusing, but he solved the problem itself, plrase read the last part of his comment:
'''
How do I delete this?
Library Manager is not Board Manager! dah..
'''

fpistm, 2017-09-14:
Thanks @Testato. I did not saw he has updated his first comment :)

closed 2017-09-14 by fpistm