stm32duino/BoardManagerFiles #12: Get the MyoBridge Library to work on stm32duino IDE

NBProjects, 2018-03-16:
Dear stm32duino team,

For a university project, I'm trying to get the MyoBridge Library by vroland (https://github.com/vroland/MyoBridge) up and running on an STM32 Nucleo F303RE Board by using your stm32duino project.
First, I programmed an Arduino UNO with the MyoBridge Library and it works fine so far. Then I installed your stm32duino package to the Arduino IDE, tried some basic programs and it also works fine.
But when I want to compile the MyoBridge (basic) projects with the stm32duino board manager, then I get several compiling errors because of missing libraries like "SoftwareSerial.h", "util/delay_basic.h" or "avr/interrupt.h".
I included these libraries manually but there comes up a new error message every time when I try to compile. Can you help me and us by get this running please?

MfG
Noah Boroske

fpistm, 2018-03-16:
Hi @NBProjects,

Thanks for trying this core.
Please, for further issue not related to package/installation issues please report to:
https://github.com/stm32duino/Arduino_Core_STM32/issues

Currently there is no port of the SoftwareSerial library as STM32 mcu's have several U(S)ART. For example, Nucleo F303RE have 5 U(S)ART available. So instead of using SoftwareSerial, I advice to use HardwareSerial anyway the lib will require some update to deal with it.
For other include, the library MyoBridge is mainly for Arduino board based on avr, I guess.
So, it will require also some update to not include those one (using ARDUINO_ARCH_STM32 as differentiator if needed).
I saw you have opened an issue to the library (https://github.com/vroland/MyoBridge/issues/9) to ask help on the porting . This is the good way, I think.
If the maintainer have some questions I could answer them.

closed 2018-03-19 by fpistm

NBProjects, 2018-03-29:
Many thanks for your answer! I will try to replace the SoftwareSerial library, for example with the HardwareSerial library. Hopefully this let compile the MyoBridge sketch.

Regards
Noah

NBProjects, 2018-05-22:
Hello,

After a little break with the project, we're now working on it again and by the replacing of the SoftwareSerial with a (Hardware) Serial, in this case Serial1, we're getting a new failure message as shown in the attached picture. Do you have an idea?
[image]

Regards
Noah

fpistm, 2018-05-22:
Hi @NBProjects ,
Please use the forum stm32duino.com for primary support.
Register and introduce you. Then you will be able to post in STM Core http://stm32duino.com/viewforum.php?f=48
Anyway, your issue is well described. Core define SUCCESS and the library too. You have to fix library to avoid redefinition.

NBProjects, 2018-06-05:
Okay, thank you, I will use the stm32duino Forum for my/our further questions.
And yes, first we only read the last few lines of the error, but with the first few lines we've got it compiled.

Regards
Noah