stm32duino/BoardManagerFiles #15: Error downloading https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

Phr3d13, 2018-11-30:
Issue is the title, link does not resolve inside arduino.ide, but appears to work when linked in normal browsing.

fpistm, 2018-11-30:
Seems to be a proxy issue. Please check Arduino proxy config

Phr3d13, 2018-12-01:
Fails for me with auto and no proxy (both should work for me, as I can click the link and view it in my web browser and it doesn't use any proxies).

fpistm, 2018-12-01:
Ok. Unfortunately, I will not able to help. It is an issue with Arduino IDE.
You could submit an issue on Arduino GitHub.

fpistm, 2018-12-01:
Maybe the same issue to this one:
https://github.com/arduino/Arduino/issues/7982

fpistm, 2018-12-17:
Referring to the Title, @Phr3d13 use the right URL.
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

pasindu-sandima, 2019-09-03:
I had the same issue and I fixed it doing the following steps.

1. Open Arduino IDE
2. Go to preferences
3. Click on the link to edit preferences.txt ( This will take you to `C:\Users\$Username\AppData\Local\Arduino15`)
4. Delete the temp file and the json file from that directory
5. Delete the package in the packages folder
6. Delete staging files related to that in the staging folder
7. Restart the device and add the link again and try

[image]
[image]
[image]