stm32duino/BoardManagerFiles #17: Consider updating the GCC varsion

ghent360, 2019-02-18:
ARM has released v8 2018-q4 of the embedded compiler. Would you consider updating the GCC version?

Is this something you host or is it coming from the Arduino folks?

fpistm, 2019-02-18:
This is something I host. 
I've plan to update it, one day. I update it when this is required to fix an issue or to support a required feature. I've updated to v6 to support arm cortex7. Anyway' I could update it soon as I know newlib has been update and fix somes possible issues. This also allow to reduce a bit the code size.

fpistm, 2019-04-03:
@ghent360 
I've released all the  packages for  `8-2018-q4-major`:
https://github.com/stm32duino/arm-none-eabi-gcc/releases/tag/8-2018-q4-major

Note that all toolchains have been rebuild due to patches required (mainly for windows).

fpistm, 2019-04-03:
Master use now the latest arm non eabi gcc toolchains:
https://github.com/stm32duino/Arduino_Core_STM32/pull/491
Will be available in 1.6.0.

closed 2019-04-03 by fpistm