stm32duino/BoardManagerFiles #22: Arduion can't down load the package of  STM32??

anymaxA, 2019-06-16:
1、when we refer this below to install the STM32 package , it failed ,
https://github.com/stm32duino/wiki/wiki/Getting-Started
2、when we check the json file of this below, the path is worng,  for the url file . there is no folder in github, it show below, can you help to change the jason file ;
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

[image]
[image]

fpistm, 2019-06-16:
Hi @anymaxA,
The url you point is correct:
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/packages/STM32-2017.5.12.tar.bz2

All old packages are available here:
https://github.com/stm32duino/BoardManagerFiles/tree/master/STM32/packages
Recent packages are available thanks the release on dedicated repo. 
Example for core version 1.5.0:
https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/1.5.0
And package:
https://github.com/stm32duino/Arduino_Core_STM32/releases/download/1.5.0/STM32-1.5.0.tar.bz2

Anyway, I've already planned to update the Wiki as several screenshots are very outdated.

anymaxA, 2019-06-16:
hi  Frederic Pillon :
        please help to check the  json file is correct, i can't downloader the  STM32 Cores by STMicroeletronics ,  it shows every slow with no speed ,can you give some 
 advice , when i try with VPN is the same;
 https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/master/STM32/package_stm_index.json

https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/packages/STM32-2017.5.12.tar.bz2
 
https://github.com/stm32duino/BoardManagerFiles/tree/master/STM32/packages
https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/1.5.0
https://github.com/stm32duino/Arduino_Core_STM32/releases/download/1.5.0/STM32-1.5.0.tar.bz2

anymaxA, 2019-06-16:
hi  Frederic:
    when copy the url to  web, it shows tree, not raw, 

 https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/packages/STM32-2017.5.12.tar.bz2
 
 https://github.com/stm32duino/BoardManagerFiles/tree/master/STM32/packages
 https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/1.5.0
 https://github.com/stm32duino/Arduino_Core_STM32/releases/download/1.5.0/STM32-1.5.0.tar.bz2

fpistm, 2019-06-16:
There is no issue.
The JSON file is correct, you probably have a proxy issue or something related to your configuration.
You can try to use the proxy settings in Arduino preferences.

> 
> hi Frederic: when copy the url to web, it shows tree, not raw,
>

Right, this is the GitHub web interface. When you click on the file it could not display it as it is a binary.
Then if you click on download you get the raw link.

anymaxA, 2019-06-16:
hi   Frederic:
     I  chekeck the net,  the proxy is auto to check ;
 and it can't downloader it, it show below, it seem not wrong address;

fpistm, 2019-06-16:
Honestly don't know how to help you.

Do you have any error reported by Arduino IDE ?
I know some people got issue when using boards manager (not only for STM32).
You can try to check if a similar issue is raised in Arduino GitHub issues or raise a new one.
https://github.com/arduino/Arduino/issues

Maybe related to your antivirus?
https://github.com/arduino/Arduino/issues/8907#issuecomment-495054504

fpistm, 2019-06-19:
I close this one as all packages url referenced in the JSON are correct.

closed 2019-06-19 by fpistm

anymaxA, 2019-06-19:
hi   Frederic:
       when i  downloder it, it show this below issue,do you have any other offline install the package file, tks;

fpistm, 2019-06-19:
What is exactly your issue ?
Host OS?
Arduino IDE version?
Error log ?

anymaxA, 2019-06-22:
hi  Frederic:
     when i down loader the STM Core by STMicroelectronics, but it can't downloader failded;
     OS:  Win7 32bit;
     Arduion:1.8.9
     Error:
 
  can you help to provide the offline file of this below and the way to install in offline;

fpistm, 2019-06-22:
You don't have anny error log?

Download each files manually: core, cmsis, gcc, tools.
Then download the JSON and change all url to point on the local path:
http://.... become file://...
Then in Arduino preferences point to your local JSON doing the same change http://....json  become file://...json
Anyway, you have an issue with Arduino IDE so  you should open an issue on Arduino github.

anymaxA, 2019-06-22:
hi  Fedredic:
   for this below file , no speed to download form github;  do you have any other way to downloader it,tks;
 See https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/1.6.0 * Tools:   See https://github.com/stm32duino/Arduino_Tools/releases/tag/1.3.0 * CMSIS:   See https://github.com/stm32duino/ArduinoModule-CMSIS/releases/tag/5.5.1 * arm-none-eabi-gcc:   See https://github.com/stm32duino/arm-none-eabi-gcc/releases/tag/8-2018-q4-major

fpistm, 2019-06-22:
Unfortunately, if your are not able to download from GitHub I could not help.

anymaxA, 2019-06-22:
hi  Frederic:
      tks of you help, tks;