stm32duino/BoardManagerFiles #34: Problems with Variant_h

nicoverduin, 2021-11-04:
HI,
I use the Sloeber-IDE 4.4 from Jantje. Is I used the old deprecated package of STM32duino 1.9.0 it works fine. Now I am using the new 2.0.0 and it seems that the variant.h is not working quite corectly. I has a line #include VARIANT_H where this variable is substituted in the right variant include file. The compiler complains that it cannot process the include statement.
If I add \" in front and at the end of the string, all works fine.

nicoverduin, 2021-11-04:
The cause is in a file called  sloeber.txt. I modified the line : 

build.info.flags=-D${build.series} -DARDUINO=${runtime.ide.version} -DARDUINO_${build.board} -DARDUINO_ARCH_${build.arch} "-DBOARD_NAME=\"${build.board}\"" -DVARIANT_H="${build.variant_h}"

into 

build.info.flags=-D${build.series} -DARDUINO=${runtime.ide.version} -DARDUINO_${build.board} -DARDUINO_ARCH_${build.arch} "-DBOARD_NAME=\"${build.board}\"" -DVARIANT_H=\""${build.variant_h}\""

Now it works like a charm
Regards
Nico

fpistm, 2021-11-04:
This is linked to sloeber so open an issue to it.

closed 2021-11-04 by fpistm