stm32duino/BoardManagerFiles #36: Error Downloading https://github.com/stm32duino/BoardManagerFiles/raw/main/STM32/package_stm_index.json

saravanan6046, 2022-01-24:
[image]

fpistm, 2022-01-24:
Hi @saravanan6046 
The json you try to  use is deprecated, you should use this one:
https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

About the issue it seems related to the new Arduino IDE version. Same file works fine with previous version.
As it is deprecated I will not try to fix this moreover there is no "<" in this file so don't know what could be wrong.
Moreover as this file is deprecated it will be removed. It was kept temporary to warn user about the deprecation.

closed 2022-01-24 by fpistm

fpistm, 2022-01-25:
Seems related to https://github.com/arduino/Arduino/issues/7982