stm32duino/BoardManagerFiles #41: Detecting libraries used... recipe.preproc.macros pattern is missing

nagarajgollar, 2022-05-15:
Hi,

i am getting below error while compiling in arduino ide

Detecting libraries used...
recipe.preproc.macros pattern is missing
Error compiling for board Generic STM32F4 series.

Regards,
Nagaraj

fpistm, 2022-05-15:
Could you be more precise on your setup?
Arduino IDE version, selected board and option, sketch, ....

nagarajgollar, 2022-05-15:
Hi,

Below are the details of my setup.

Board ; STM32F401

Arduino IDE version: 1.8.19

Sketch : Blink led from the example

[image]

Regards,
Nagaraj

fpistm, 2022-05-15:
Thank you.
Seems obvious as it works properly on our side.
I think you have an issue in your setup.
You should try om a fresh install to see if you reproduce it.

fpistm, 2022-05-16:
Could you tell us which core version you used ? And where you get it ?
As far as I can see, all people having this error had a wrong setup.

nagarajgollar, 2022-05-16:
Hi ,

i have used below examples
[image]

Regards,
Nagaraj

fpistm, 2022-05-16:
This is library. I ask you the core version and which json file you used?
And try a clean installation please.

nagarajgollar, 2022-05-16:
Hi,

I have used the below json file
https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/main/package_stmicroelectronics_index.json

Arduino 1.9.19

Regards,
Nagaraj

fpistm, 2022-05-16:
I guess it is 1.8.19.  Json file is the correct one.
All seems correct. Please try to reinstall in a fresh install to be sure all is fine.
This error is not linked to the core as it works well with this setup. You can also refer to the CI which is passed. I have no more clue for this. If you reproduce it you should open an issue on arduino ide GitHub.