stm32duino/BoardManagerFiles #48: Error in compiling the code in arduino ide for stm32f03 board, a big thanks if anyone can help

hakam007, 2022-09-29:
hello sir , i am facing a error problem since more than a month in arduino IDE ,i am using stm32f103 board, error is as given below:
arm-none-eabi-g++: error: Singh\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0=.: No such file or directory
exit status 1
Error compiling for board Generic STM32F1 series.

then i puchased a new board stm32f401 board ,then also found same error ... i really dont know what i can do ,kindly help me in this it will be a really a great help... thanks
[image]

fpistm, 2022-09-29:
Hi, it is already know and fixed thanks https://github.com/stm32duino/Arduino_Core_STM32/pull/1759

Fix will be available in the 2.40 release.
Simply replace C:\Users\<your username>\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0\system\extras\prebuild.sh by this file while 2.4.0 is not released:
https://github.com/stm32duino/Arduino_Core_STM32/blob/main/system/extras/prebuild.sh

Duplicate of 
https://github.com/stm32duino/Arduino_Core_STM32/issues/1829

closed 2022-09-29 by fpistm

hakam007, 2022-09-29:
@fpistm  oh my God  you are such a genuis person ,really no words to thankyou...you proved saviour for me.. may god bless you

hakam007, 2022-09-29:
woww so nice.. i am feeling like m talking to a great person

hakam007, 2022-09-30:
@fpistm  Hii sir , i am facing problem in uploading code in stm32f103c8t6 , uploading error occured.  plz guide me in this
[image]

fpistm, 2022-09-30:
Hi @hakam007 

please use the forum for such support: http://www.stm32duino.com