stm32duino/BoardManagerFiles #50: Error downloading the board manager package in Arduino IDE 1.8.19

lufu007, 2022-10-02:
Hi there,

Following the instructions I fail at the point of searching in the board manager for the stm32duino as the following error occurs:

Error downloading https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json
Error downloading https://downloads.arduino.cc/packages/package_index.json

I made sure I don't have any proxy connection. 
I would appreciate to receive help if the problem lies on my version of the Arduino IDE 1.8.19 or is a problem of other origin.
Thanks in advance.

lufu007, 2022-10-02:
I added the wrong error message. I also tried the new link but it doesn't work either.

Error downloading https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json
Error downloading https://downloads.arduino.cc/packages/package_index.json

fpistm, 2022-10-02:
Seems you have network issue as even the official arduino boards index could not be downloaded. Check your connection and network proxy configuration in the Arduino IDE preferences.

closed 2022-10-02 by fpistm