stm32duino/BoardManagerFiles #51: The link provided is not working. I'm not able to download the boards manager for stm32

dhruv009dj, 2022-12-26:


rohit8vmu, 2022-12-26:
Yes, I am also facing the same issue. I am using Arduino 1.8.19. The Board Manager link I am using in the preference is:
https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json

This shows an error in downloading the package from the link.
I wonder what could be the correct JSON link?

fpistm, 2022-12-26:
main not master:
https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

closed 2022-12-26 by fpistm

dhruv009dj, 2022-12-27:
@fpistm I have used no proxy in my settings. But the link is still not working.

fpistm, 2022-12-27:
File is available on GitHub so it is valid. 
If you used a browser you should be able to download it.
Try to run arduino in debug mode or from a console to see if you have more info. 
Anyway I have no more idea as file and link are valid. I bet on Arduino or environment issue.

fpistm, 2022-12-27:
And you only told that link is not valid. To get more support you should described precisely your setup, link you used,...