stm32duino/BoardManagerFiles #55: Legacy package indexes removed

matthijskooijman, 2023-04-14:
Hey @fpistm, you removed the legacy package index (e.g. https://github.com/stm32duino/BoardManagerFiles/raw/dev/STM32/package_stm_index.json), I was still using that :-) We're still using a fork of the core based on 1.9.0, so I need to install your (unmodified) version to get the right tools to use our fork...

Any chance you could restore the index (preferably at the original URL)?

fpistm, 2023-04-17:
Hi @matthijskooijman 
you can use the master branch which kept legacy files.
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

I removed it from the main to avoid mistake (already had several feedback about this)

matthijskooijman, 2023-04-24:
Works like a charm, thanks!