stm32duino/BoardManagerFiles #56: problem installing STM32 cores to arduino's Board manager

AnilMudulakar, 2023-06-01:
Greetings of the day,
Am using Arduino 2.1.0 IDE and facing problem installing STM32 to boards manager, and iam getting this error even after lot of trials

`Some indexes could not be updated. Get "https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/main/package_stmicroelectronics_index.json": dial tcp [2405:200:1607:2820:41::36]:443: connectex: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.: https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json `

Any help will be appreciated
Thanks

fpistm, 2023-06-01:
Hi @AnilMudulakar 
Unfortunately I can't help on this. It is linked to Arduino IDE, and/or GitHub issue and/or Internet connection.

closed 2023-06-01 by fpistm

AnilMudulakar, 2023-06-01:
thank u