stm32duino/BoardManagerFiles #61: tar.bz2 files hosted by github are extremely slow to download, what is prevents installation

Pethical, 2023-12-07:
Something happened in the last few days and the package simply cannot be installed, the download does not finish even after 2-3 hours. Cloning the repo works fine, the download of the zip files under the release is fast, however, the bz2 files are also extremely slow to download via the web, 10kb/s speed is reached for some reason. I'm well aware that this is a GitHub issue, however, since the package is only available here it completely prevents installation.

I seen a lot of issues beacause the same problem, eg: #56 #58 #60 

Maybe it's worth to find an alternative download location, e.g. GitLab, Google Drive, or anything from which the download works more stably than from GitHub, or use something else instead of bz2 files, if the Arduino IDE supports it, because apparently there is a correlation between the type of file and the download speed of GitHub, at least from here in Europe.

fpistm, 2023-12-07:
Hi @Pethical 
The issue you mentioned is not for the archive but the json file. So no link. First time I heard issue about the archive and/or its extension. Maintaining different storage is not a solution as it would mean to maintain several json which could lead to Arduino package index issue. About bz2 extension if you have any official input I will investigate. 
At this time I will not change anything. More than 400k download since several years and first time this kind of issue raised. So close it.

closed 2023-12-07 by fpistm

Pethical, 2023-12-07:
Thank you for the answer! I didn't have a problem with this before either.
In this case, I will permanently switch to PlatformIO.
Thanks again!