stm32duino/BoardManagerFiles #67: GCC update to 13

dojyorin, 2024-05-16:
Could you update GCC version from current 12.2.1 to 13.2.1?
- https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/tag/v13.2.1-1.1

This update may also resolve previously posted issue.

- https://github.com/stm32duino/Arduino_Core_STM32/issues/2138
- http://nemuisan.blog.bai.ne.jp?eid=244160 (Japanese page)

fpistm, 2024-05-16:
Hi @dojyorin 
This is issue is linked to the core not to this repo.
It is already updated to a 13.2.1
https://github.com/stm32duino/Arduino_Core_STM32/pull/2315
This will come within next release of the core planne in the coming weeks.

closed 2024-05-16 by fpistm

dojyorin, 2024-05-28:
Thank you for resolving issue.
I'm always grateful for your help :)