stm32duino/BoardManagerFiles #68: How to modify the upload address when using DFU mode?

TangYanYee, 2024-06-17:
-------------------------------------------------------------------
                       STM32CubeProgrammer v2.15.0                  
      -------------------------------------------------------------------

USB speed   : Full Speed (12MBit/s)
Manuf. ID   : STMicroelectronics
Product ID  : STM32 DownLoad Firmware Update
SN          : 205335724653
DFU protocol: 1.1
Board       : --
IAP

Memory Programming ...
Opening and parsing file: stm32mb.ino.bin
  File          : stm32mb.ino.bin
  Size          : 33.76 KB 
  Address       : 0x08000000 
  
  Right now the address is at 0x08000000, may I ask how should i change it to other address?

fpistm, 2024-06-17:
Hi @TangYanYee 
First, do not open issue for question/support, [forum](stm32duino.com) or [discussions](https://github.com/orgs/stm32duino/discussions) are better place for this. Moreover it is not link to the BoardManagerFiles GitHub repository.

You have to specify the `build.flash_offset` linked to your board using the [boards.local.txt](https://arduino.github.io/arduino-cli/0.34/platform-specification/#boardslocaltxt)

closed 2024-06-17 by fpistm

TangYanYee, 2024-06-21:
ok, thanks for your help. Sorry if I've caused any inconvenience.