stm32duino/BoardManagerFiles #69: Adding this JSON file to Additional Boards URL breaks the "Arduino IDE Tools | Ports" dropdown

billbrach, 2024-07-06:
Additional boards URL that I have in my ".arduino15/preferences.txt" file, is the existing one used for these boards.
It is: https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

Software/Hardware:

Arduino 1.8.19 running on:

System:    Kernel: 5.4.0-187-generic x86_64 bits: 64 compiler: gcc v: 9.4.0 
           Desktop: Cinnamon 4.6.7 wm: muffin dm: LightDM Distro: Linux Mint 20 Ulyana 
           base: Ubuntu 20.04 focal 
Machine:   Type: Desktop System: Dell product: OptiPlex 755 v: N/A serial: <filter> Chassis: 
           type: 15 serial: <filter> 
           Mobo: Dell model: 0PU052 serial: <filter> BIOS: Dell v: A22 date: 06/11/2012 
CPU:       Topology: Dual Core model: Intel Core2 Duo E6550 bits: 64 type: MCP arch: Core Merom 
           rev: B L2 cache: 4096 KiB 
           flags: lm nx pae sse sse2 sse3 ssse3 vmx bogomips: 9309 
           Speed: 1995 MHz min/max: 2000/2333 MHz Core speeds (MHz): 1: 1995 2: 1995 
Graphics:  Device-1: AMD Cedar [Radeon HD 5000/6000/7350/8350 Series] vendor: Dell driver: radeon 
           v: kernel bus ID: 01:00.0 chip ID: 1002:68f9 
           Display: x11 server: X.Org 1.20.13 driver: ati,radeon unloaded: fbdev,modesetting,vesa 
           resolution: 1680x1050~60Hz 
           OpenGL: renderer: AMD CEDAR (DRM 2.50.0 / 5.4.0-187-generic LLVM 12.0.0) 
           v: 3.3 Mesa 21.2.6 compat-v: 3.1 direct render: Yes 

**Describe the bug**
Cannot select any USB ports in the Tools | Ports dropdown because it is greyed out.  This was working correctly until sometime near the end of June 2024.

**To Reproduce**
On MY system, if I add your JSON url to the Arduino IDE Files | Preferences/Additional Boards Manager URL, the Arduino IDE Tools | Ports dropdown is greyed out.  If I remove it, I can then select a Port.

It is entirely possible there is some other problem with my system but thought I ought to report it, in case something has changed in either the OS, Java, or the Arduino IDE.  I don't understand how your Add'l Boards URL would affect these items but hoping you will have a solution. I tried with *only* your JSON file added to the Add'l Boards URL, and it still broke the Tools | Ports dropdown.

If I start the IDE from the command line, this is the error message I get.  The "Picked up JAVA_TOOL_OPTIONS: " message is normal, for a working startup from the command line,  the "Exception ..." is extra on the broken version.  If I run lsusb from the command line, and plug/unplug a board, I see what looks to be correct.  Doing this with dmesg at the command line, also looks to be correct.

(command line startup message)
billbrach@LinuxMint20:~/arduino-1.8.19$ ./arduino
Picked up JAVA_TOOL_OPTIONS: 
Exception in thread "cc.arduino.packages.discoverers.serial.SerialDiscovery" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.LinkedList.checkElementIndex(LinkedList.java:555)
	at java.util.LinkedList.get(LinkedList.java:476)
	at processing.app.Platform.resolveDeviceByVendorIdProductId(Platform.java:188)
	at cc.arduino.packages.discoverers.serial.SerialDiscovery.forceRefresh(SerialDiscovery.java:166)
	at cc.arduino.packages.discoverers.serial.SerialDiscovery$1.run(SerialDiscovery.java:96)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)

fpistm, 2024-07-06:
Just having the json in the arduino preferences ?
Or you select an stm32 board?
Arduino 1.8.x is no more supported sinced 2.8.0 due to several changes in the arduino specifications.

billbrach, 2024-07-06:
Just having in the Additional Boards Manager URL list.  In fact, even if it is only added URL, it does as described - greyed out Ports dropdown.  Could your URL be modifying the Arduino IDE tools path somehow ?? The usual issue with a greyed out Ports dropdown is permissions, but this system has been running for 3 years w/o issue.  I'm pretty sure this started either because of a Linux Mint update or *possibly* when the IDE was open, but the computer locked up on a Suspend, and I had to hard boot it.  Some AppImage files will not Suspend properly on this computer, which is probably an AppImage issue.  Entirely possible your URL is fine and something else is corrupt.  I need to completely remove your package that exists on my machine, and try again, as it's possible that is where the corruption is.  I'll try to do that in the next day or two.  And THANKS for your prompt reply !!

fpistm, 2024-07-06:
Or Arduino cache issue corrupted. Unfortunately I could do nothing. It is an  an Arduino issue.

billbrach, 2024-07-06:
Thanks for looking at it.  Like I said, I have no idea where the issue is.  If I figure it out, I will post back.

fpistm, 2024-07-08:
OK. I've reproduced.
Since 2.8.0, the new arduino-cli specification allows to define several vid.x / pid.x and it seems not supported by Legacy Arduino 1.8.19.
```
Exception in thread "cc.arduino.packages.discoverers.serial.SerialDiscovery" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
	at java.util.LinkedList.checkElementIndex(LinkedList.java:555)
	at java.util.LinkedList.get(LinkedList.java:476)
	at processing.app.Platform.resolveDeviceByVendorIdProductId(Platform.java:188)
	at cc.arduino.packages.discoverers.serial.SerialDiscovery.forceRefresh(SerialDiscovery.java:166)
	at cc.arduino.packages.discoverers.serial.SerialDiscovery$1.run(SerialDiscovery.java:96)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
```
As we do not provide a SerialDiscovery (this is not a bug :wink;) it seems not able to resolve COM port.

As stated Legacy Arduino ide 1.8.x is no more supported due to those breaking specifications changes so can't do nothing for this.

closed 2024-07-08 by fpistm

billbrach, 2024-07-09:
Interestingly, I found a google hit on how to use Arduino's serial-discovery tool, to test the USB ports.  That program behaved exactly as it was supposed to.  Plug a board in, and it would see it, unplug a board and it would see and report that too.

You asked in an earlier message, about Arduino 1.8.X being deprecated.  The *only* reason I use it is because it is easier to use than 2.X, and more importantly, it compiles faster.